GeekMagicClock/smalltv #1: Could you share the source code? 🤗

mowtschan, 2023-09-20:


Cabalist, 2023-09-20:
I would also love to see the source.   Having an open-source firmware would have me buying a dozen of these.

mowtschan, 2023-09-27:
just found another very interesting project: https://github.com/Xinyuan-LilyGO/T-QT

GeekMagicClock, 2023-09-29:
Hi, glad you like it, but as for some reason we can't share the code for the present, commercial copies would bother us a lot, but i was think about offer some basic code for DIYers.

Thumok, 2023-12-30:
Would love to have a basic version for some DIY stuff.
I'd love to have a clock page with weather that doesn't scroll and is less distracting that the default one.

Eziokiller470, 2024-01-12:
I love this little gadget. One thing I'd have liked is add my own dht22 sensor inside for indoor temperature/humidity. I see there are free pins inside that would work.

Would need to adapt the code to display those data instead of the animated GIF.

Could you maybe keep the firmware locked but add an option to add more codes to run? Thanks.

GeekMagicClock, 2024-01-13:
> I love this little gadget. One thing I'd have liked is add my own dht22 sensor inside for indoor temperature/humidity. I see there are free pins inside that would work.
>
> Would need to adapt the code [...]

Hi, we have that idea before and tested with dht sensor, but the space is quite limited inside the box, the board and screen heat will cause the data not accurate.

But I'm planning to start an open source project, with extra free gpio pins on board, inside the same box, some basic display code, for the developer to fit their needs.

LoneDev6, 2024-04-30:
Would be nice to get the source code of this in order to edit a few things and possibly contribute to its development.
If your fear is the risk of getting your device copied I guess people can already copy your board and install your bin file without having access to the source.

So please consider this. People will help you coding updates for free.