GeekMagicClock/smalltv #2: After upload new firmware, unable to access the device

exampleex, 2023-09-24:
After successfully updating the device with firmware version 3.1.1, I encountered an issue. Although the uploading process completed without any problems, I am unable to access the device's address. 
A 404 error appears when I attempt to access. 

The device itself appears to be functioning properly.

exampleex, 2023-09-24:
[image]

GeekMagicClock, 2023-09-29:
Hi, earlier version before 3.1.0 can only be updated to 3.0.33, please check this.

exampleex, 2023-09-29:
> Hi, earlier version before 3.1.0 can only be updated to 3.0.33, please check this.

So the older version won't be able to update anymore, right?

GeekMagicClock, 2023-09-29:
> > Hi, earlier version before 3.1.0 can only be updated to 3.0.33, please check this.
> 
> So the older version won't be able to update anymore, right?

As for different software structure, little rom space in earlier version for use, only bug fixes are possible, but new features would be added to v3.1.0 and later versions.