GeekMagicClock/smalltv #3: How can I update this device firmware?

samueljake, 2023-09-26:
[image]
I tried many times.
I can not connect my device's ip address/update.

Cabalist, 2023-09-28:
I'm also having this issue.

GeekMagicClock, 2023-09-29:
Hi, seems the response is not from the clock, what version do you have?

exampleex, 2023-09-29:
> [image] I tried many times. I can not connect my device's ip address/update.

> I'm also having this issue.

Try to access from the front page.
[image]

GeekMagicClock, 2023-09-29:
> > [image] I tried many times. I can not connect my device's ip address/update.
> 
> > I'm also having this issue.
> 
> Try to access from the front page. [image]

what verison do you have?

exampleex, 2023-09-29:
> what verison do you have?

3.0.33

GeekMagicClock, 2023-09-29:
1. For earlier versions, v3.0.x etc, please update to 3.0.33, that's stable and the latest, do not update to v3.1.0 or later, or the web pages will be not found.
2. For versions after v3.1.0, could be updated to v3.1.x.

GeekMagicClock, 2023-09-29:
> > what verison do you have?
> 
> 3.0.33

What's the version before you update it to v3.0.33? I have tested v3.0.33, no such problem appeared

exampleex, 2023-09-29:
> > > what verison do you have?
> > 
> > 
> > 3.0.33
> 
> What's the version before you update it to v3.0.33? I have tested v3.0.33, no such problem appeared

3.1.1

nikopiko666, 2023-11-23:
> 1. For earlier versions, v3.0.x etc, please update to 3.0.33, that's stable and the latest, do not update to v3.1.0 or later, or the web pages will be not found.
> 2. For versions after v3.1.0, could be updated to v3.1.x.

how to update the firmware whe /update doesnt work?

luhis, 2024-08-11:
I have v2.5.4, how do i update the firmware? soldering on a serial adapter is fine

closed 2024-10-09 by GeekMagicClock