GeekMagicClock/smalltv #4: Hardware info

breti, 2023-09-28:
Hello,

smalltv and smalltv pro are interesting little devices :-). Thank you for having already prepared the ISP header so that it is easy to install a different firmware.

We are currently trying to develop an [ESPhome configuration](https://community.home-assistant.io/t/installing-esphome-on-geekmagic-smart-weather-clock-smalltv-pro) for the devices so that they can be used as a home assistant display. Unfortunately, we have not yet managed to bring the display to life. Would you be willing to tell us a little more about the display and the GPIO ports used?

Thanks a lot for your help! :-)

GeekMagicClock, 2023-09-29:
Hi, here for your reference.
#define TFT_BL   5            // LED back-light control pin
#define TFT_BACKLIGHT_ON LOW  // Level to turn ON back-light (HIGH or LOW)
#define TFT_CS   -1  // Chip select control pin D8
#define TFT_DC   0  // Data Command control pin
#define TFT_RST  2  // Reset pin (could connect to NodeMCU RST, see next line)
SCK 14
MOSI 13

breti, 2023-09-29:
Thanks a lot :-) So the pinout we figured out is correct:

|1| GND
|2| GP0 → D/C
|3| GND → CS ? tied to GND
|4| GP14 → SCK
|5| GP13 → MOSI
|6| GP2 - > reset
|7| VCC
|8| GND
|9| GP5 → Back-light Anode through P MOSFet
|10| GND → Back-light Cathode

What's the type of the display controller? We've tried st7789, ILI9341, ILI9342, but nothing worked.

GeekMagicClock, 2023-09-29:
> st7789

st7789

breti, 2023-10-04:
Can you please add the GPIO assignments of the small tv pro / ESP32 version?

hitesh-singh, 2023-11-22:
> Can you please add the GPIO assignments of the small tv pro / ESP32 version?

+1