GeekMagicClock/smalltv #5: Updating firmware on a different build of the weather clock

gquiring, 2023-10-29:
It looks like there are different source versions for this clock.  My SSID shows up as Smart Weather Clock not GIFTV.  When I enter AP mode the web page has no option to upload firmware.  There seems to be no support for my version, I contacted the seller at AliExpress but got no response.  Is there a way to upload your firmware via some other method?  I have the Arduino IDE installed on my PC, could that be used to upload the bin file somehow?  

My main issue is the city location, my version does not allow an ID to be input.  So I cannot select the proper city, as my city exists in multiple areas in the USA. 

This is what my web page looks like:

[image]

GeekMagicClock, 2023-11-03:
That is copied version, not the orginal from us.

rzboril, 2024-01-12:
Yes i upgraded from owner google disc firmware.
If you still have a problem, write and I will send you the procedure.

GeekMagicClock, 2024-01-13:
Please do not buy fake or copied product, please support GeekMagic original products.

closed 2024-01-13 by GeekMagicClock

rzboril, 2024-01-14:
Yes i upgraded original product.

nacon03, 2024-08-14:
> Yes i upgraded from owner google disc firmware. If you still have a problem, write and I will send you the procedure.

Hey how did you do it? I have the same problem :(