GeekMagicClock/smalltv #12: Configuration screen has less features after update

AllanKley, 2023-12-14:
I was having some problems with the weather so I decided to update the firmware to the latest version. That ended up causing more problems so I decide to go for the V3.0.33 to see if it would work and it did. The only problem is that the configuration screen before all the updates had more features (I could for example go to 192.168.18.211/images and configure my images to cycle between them), now I can only select a one from a couple of gifs and they are not even appering

GeekMagicClock, 2023-12-15:
Hi, recently the default public weather api key is run out, just need to update the key, you could register for your own, it's free.
https://home.openweathermap.org/api_keys
And you have to update the correct firmware for that model you have.

AllanKley, 2023-12-15:
I have already registered a new api key, the weather is working properly and even the little gif on the bottom right works if I upload it myself. The only thing is that before I put it on v3.0.33 I was able to switch modes, for example forecast, different clock faces and even just a jpg but now on v3.0.33 I can only have it on the one in the image. Is there a way for me to see which firmwares are correct for my device?
[image]
[image]

GeekMagicClock, 2023-12-15:
According what you said, the features are for smalltv-ultra, so you need to update the correct firmware, check the repo[ here](https://github.com/GeekMagicClock/smalltv-ultra/tree/main/V9.0.11).

AllanKley, 2023-12-15:
All right, thanks for the help

closed 2023-12-15 by AllanKley