GeekMagicClock/smalltv #14: unable to get to the weebpage after the update !!

ssama3, 2023-12-17:
the clock seems to be working just fine, but i am unable to get to the webserver page. after the update.

❯ ping 10.0.0.26
PING 10.0.0.26 (10.0.0.26): 56 data bytes
64 bytes from 10.0.0.26: icmp_seq=0 ttl=255 time=34.118 ms
64 bytes from 10.0.0.26: icmp_seq=1 ttl=255 time=253.079 ms
64 bytes from 10.0.0.26: icmp_seq=2 ttl=255 time=120.277 ms
64 bytes from 10.0.0.26: icmp_seq=3 ttl=255 time=8.270 ms
64 bytes from 10.0.0.26: icmp_seq=4 ttl=255 time=51.982 ms
64 bytes from 10.0.0.26: icmp_seq=5 ttl=255 time=177.312 ms
64 bytes from 10.0.0.26: icmp_seq=6 ttl=255 time=96.330 ms
64 bytes from 10.0.0.26: icmp_seq=7 ttl=255 time=248.919 ms
^C
--- 10.0.0.26 ping statistics ---
8 packets transmitted, 8 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 8.270/123.786/253.079/88.421 ms

[image]

ssama3, 2023-12-17:
is there a way i can repush the bin file using ssh or other commands ?

GeekMagicClock, 2023-12-17:
You can try http://10.0.0.26/update, and then update back.

closed 2024-10-09 by GeekMagicClock