GeekMagicClock/smalltv #15: HTTP custom text and images

mungle, 2023-12-20:
Hello!
I’d like to display some text messages and images via http requests using nodered. As you don’t distribute the source code, can you implement similar features for the “basic” version?
Thank you!

GeekMagicClock, 2023-12-21:
Hello, this is possible, but it also depends on your needs, how big the font size and layout style you need, maybe a simple interface can be provided.

mungle, 2023-12-21:
If it can be displayed an html, with a simple css style.

With a HTTP request like:
http://192.168.0.2/message?title=hello&message=world

and a html section in the web admin like:

```
<h1 class=“uppercase”>%title%</h1>
<p class=“red”>%message%</p>
```

and a css section with:

```
.uppercase{ text-transform: uppercase; }
.red{ color: red; }
```

the message can be easily customized! There are also other and better methods I believe (JSON?).
It would be great to display also images sent by HTTP requests! 👍👍👍

This would be very appreciated by me 🤩 and many, many users, I believe!!!

GeekMagicClock, 2023-12-23:
Okay, let's vote & talk.

GeekMagicClock, 2023-12-23:
Is it some thing like this?
[image]
Title section: font size, font color.
Message section: font size, font color.
Image section: image size.

It's better to fix the font-size and image size, since only a few font-size is designed in the firmware for the present.
And I wonder what's that used for, what kind of situation.

mungle, 2023-12-23:
Hello,
I'd use to send home notifications like, reminders, appointments or post-it messages.
I'd also like to use to view custom image screenshot from my security camera system when a motion alert is detected (I'm now using a telegram bot and it's super useful!!!)

To be as flexible as possible I suggest four style:

### **1 - Text only**
[image]

### **2 - Text + custom icon**
[image]
[image]
[image]

### **3 - Text + background image**
[image]
_Credits: Unsplash_

### **4 - Image only**
[image]
_Credits: Unsplash_
[image]

The display style change based on the HTTP message. If there's message=[text] only, the smalltv will ignore the other inputs (title, icon and image) as they're "empty" displaying the message field only.
 
Once a new message is sent, the previous one will be deleted.

The image only can be the most flexible solution as you can generate with an external script an image with any content in any style and layout you like mixing text and images.

GeekMagicClock, 2023-12-27:
Hi, your PS skills is great. 

Anyone interested in this please leave your message.

mungle, 2024-01-28:
Hello! Do you have updates on this?

PhantomArt, 2024-02-03:
I would like to be able to display a custom 240x240 image via the API, no animation needed, just a static image. Then I could create an image with my application with any information and any design, and use the device only as a display to display this image.

palecour, 2024-03-20:
👍🏻

DanHKfree97, 2024-03-20:
Hi, im here for vote! Would be geat function... :)

Martinho672, 2024-03-29:
I will love that function too

GeekMagicClock, 2024-03-30:
> Hello! Do you have updates on this?

Hello, I'm going to add the image full screen update feature.

Andre-Lehrmann, 2024-04-15:
Hello,
i am always interested in this function.
Have a nice day.

pmout, 2024-04-19:
Hello, GeekMagicClock
Can you share if it is possible to connect to the device via SSH and how to do it?

mungle, 2024-04-27:
> > Hello! Do you have updates on this?
> 
> Hello, I'm going to add the image full screen update feature.

I’m super excited as many others I believe. Do you have any ETA for the feature?

4mitabh, 2024-05-25:
This will be a great addition.
Please make this happen!

gholtike, 2024-05-28:
I'm also here to vote, I'm excited about the idea, I think it can meet the needs of many different applications.

4mitabh, 2024-05-30:
Or if we can configure an HTTP API what the device can poll, and update the screen with the data returned (in a predefined schema that the device supports)  by the API.

donatellosantoro, 2024-06-04:
Hi there
+1

Similar to https://github.com/GeekMagicClock/smalltv-pro/issues/42

callumj, 2024-06-06:
If this was supported: I’d be ordering at least 5. I have a need to display custom notifications as well as stats from my weather station or motion sensors.

ijavid, 2024-06-07:
Voting here +1

gholtike, 2024-06-15:
Pleeeease

4mitabh, 2024-06-16:
Here is a proposal for API support what will greatly help users who want to display their own data on this cute little device.

1. We add a new ‘Theme’ for API along with the existing ones. 
Like, in the screen below:
[image]

2. Users can go to the API tab and configure their own API URLs. They can also configure the time in seconds between subsequent API polls and updates. This is pretty similar to what we have on the ‘Photo Albums’ tab.

[image]

 
Once the user has added their API URLs, the device will periodically query these APIs, fetch the data, and automatically display it on the device's screen. This would provide users with a powerful and flexible way to display their own data on the device, tailoring the content and update frequency to their specific needs.

The APIs must return the data in JSON format, following a standardized structure that the device can easily parse and display. We can publish this JSON schema on smalltv github page.

I propose the following schema for the JSON API response structure. This is based on the [beautiful design mockups](#issuecomment-1868316202) that have been shared previously by @mugle.

**Text Only:**

```
{
  "layout": 1,
  "title": "Lorem Ipsum",
  "text": "Lore  ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
}
```

**Text and Icon:**

```
{
  "layout": 2,
  "icon": "🔔",
  "title": "Lorem Ipsum",
  "text": "Lore  ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
}
```

**Text and background Image**

```
{
  "layout": 3,
  "image": "http://url.to/image",
  "text": "Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
}
```
Image Only:
```
{
  "layout": 4,
  "image": "http://url.to/image"
}
```

mattkri, 2024-06-20:
Voting here for the Smalltv-ultra!

mattkri, 2024-06-24:
I know this isn't a great solution and kind of hacky, but if you want to just have a script monitor a folder for a jpeg and upload it, I wrote this script in python that will monitor a directory that you set (in the bottom) for any jpg, and then resize, and upload that jpg to your device, while deleting the previous image.

Just set your device's IP address, and the correct folders to watch.

You could modify this to use in whatever setup you want.

```
import os
import time
import requests
from watchdog.observers import Observer
from watchdog.events import FileSystemEventHandler
from PIL import Image, ImageOps
import shutil

class ImageHandler(FileSystemEventHandler):
    def __init__(self, upload_url, delete_url, filename, temp_dir):
        self.upload_url = upload_url
        self.delete_url = delete_url
        self.filename = filename
        self.temp_dir = temp_dir
        os.makedirs(temp_dir, exist_ok=True)

    def on_modified(self, event):
        if event.src_path.endswith(".jpg") or event.src_path.endswith(".jpeg"):
            print(f"Detected change in {event.src_path}. Processing...")
            temp_image_path = self.resize_and_pad_image(event.src_path)
            self.delete_previous_image()
            self.upload_image(temp_image_path)

    def resize_and_pad_image(self, image_path):
        with Image.open(image_path) as img:
            img = ImageOps.pad(img, (240, 240), color="black")
            temp_image_path = os.path.join(self.temp_dir, self.filename)
            img.save(temp_image_path)
            print(f"Image resized and padded to 240x240 pixels: {temp_image_path}")
        return temp_image_path

    def delete_previous_image(self):
        response = requests.get(self.delete_url)
        print(f"Deleting previous image at {self.delete_url}")
        print(f"Response status code: {response.status_code}")
        if response.status_code == 200:
            print("Deletion successful")
        else:
            print(f"Deletion failed with status code {response.status_code}")

    def upload_image(self, image_path):
        with open(image_path, 'rb') as image_file:
            files = {'file': (self.filename, image_file)}
            response = requests.post(self.upload_url, files=files)
            print(f"Uploading to {self.upload_url} with file {self.filename}")
            print(f"Response status code: {response.status_code}")
            print(f"Response text: {response.text}")
            if response.status_code == 200:
                print("Upload successful")
            else:
                print(f"Upload failed with status code {response.status_code}")
                print(f"Response headers: {response.headers}")
                print(f"Response content: {response.content}")

def monitor_directory(directory_path, upload_url, delete_url, filename, temp_dir):
    event_handler = ImageHandler(upload_url, delete_url, filename, temp_dir)
    observer = Observer()
    observer.schedule(event_handler, directory_path, recursive=False)
    observer.start()
    try:
        while True:
            time.sleep(1)
    except KeyboardInterrupt:
        observer.stop()
    observer.join()

if __name__ == "__main__":
    # Use raw string to handle backslashes in Windows paths
    directory_to_watch = r"D:\My Documents\Picture Watchdog\Picture"
    upload_endpoint = "http://192.168.0.172/doUpload?dir=/image/"
    delete_endpoint = "http://192.168.0.172/delete?file=%2Fimage%2F%2Fimage.jpg"
    image_filename = "image.jpg"
    temp_directory = r"D:\My Documents\Picture Watchdog\Picture\temp"
    monitor_directory(directory_to_watch, upload_endpoint, delete_endpoint, image_filename, temp_directory)

```

callumj, 2024-06-26:
Building on @mattkri what I am doing for now is using an image library in my preferred programming language (Go) to build an image with my preferred data that I want to display and then uploading the image inside my program.

4mitabh, 2024-06-27:
My worry is that this solution might kill the flash with too many writes, if we are updating the images frequently. That's why, an API that fetches and displays text would have been really great.

philipag, 2024-07-01:
+1 for smalltv-pro

Phara0h, 2024-08-13:
Would love this, so can add custom stuff to the screen, or better yet open source this project and let us help you for free

niallb-ie, 2024-08-20:
+1 from me too. Currently using the pro.

aspyk, 2024-09-18:
+1
custom text or image display from API would be a killer feature for me !

someth, 2024-09-24:
> > Hello! Do you have updates on this?
> 
> Hello, I'm going to add the image full screen update feature.

Did you add this feature to smalltv firmware yet?

jafuuu, 2024-10-12:
> Here is a proposal for API support what will greatly help users who want to display their own data on this cute little device.
> 
>     1. We add a new ‘Theme’ for API along with the existing ones.
>        Like, in the screen below:
>        ![1](https://private-user-images.githubusercontent.com/1475058/340138123-501255c9-4422-435a-b9d1-330badc5b1c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3NzYyMzksIm5iZiI6MTcyODc3NTkzOSwicGF0aCI6Ii8xNDc1MDU4LzM0MDEzODEyMy01MDEyNTVjOS00NDIyLTQzNWEtYjlkMS0zMzBiYWRjNWIxYzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTJUMjMzMjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjY1ODk5OWFkOWZjZmY5MmY2ZWRlNjgzYmVlMTNmOGVlNjE3N2ZkY2YzZDlmYzVkZDM0ZTNkNjA3MWY4YWZlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.VreeNubV2uhzRK1W5-jiafZ6eVpK2SX8raW4CFRHmw0)
> 
>     2. Users can go to the API tab and configure their own API URLs. They can also configure the time in seconds between subsequent API polls and updates. This is pretty similar to what we have on the ‘Photo Albums’ tab.
> 
> 
> ![2](https://private-user-images.githubusercontent.com/1475058/340138313-8a37ce70-427f-4d56-af1d-108ef4580022.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3NzYyMzksIm5iZiI6MTcyODc3NTkzOSwicGF0aCI6Ii8xNDc1MDU4LzM0MDEzODMxMy04YTM3Y2U3MC00MjdmLTRkNTYtYWYxZC0xMDhlZjQ1ODAwMjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTJUMjMzMjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTIzMWNmNzNjZWE3MTE2NTlkNTFhOTZkYzNjNTlhNGYxMjMzMTNiODcxOGY0NzE1ZmY5YzNiODljOTIxMGY4YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.l7UarrLcHrKZFgp1KktvwDQc5QeTJDv-6dBCRvBs-a4)
> 
> Once the user has added their API URLs, the device will periodically query these APIs, fetch the data, and automatically display it on the device's screen. This would provide users with a powerful and flexible way to display their own data on the device, tailoring the content and update frequency to their specific needs.
> 
> The APIs must return the data in JSON format, following a standardized structure that the device can easily parse and display. We can publish this JSON schema on smalltv github page.
> 
> I propose the following schema for the JSON API response structure. This is based on the [beautiful design mockups](#issuecomment-1868316202) that have been shared previously by @mugle.
> 
> **Text Only:**
> 
> ```
> {
>   "layout": 1,
>   "title": "Lorem Ipsum",
>   "text": "Lore  ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
> }
> ```
> 
> **Text and Icon:**
> 
> ```
> {
>   "layout": 2,
>   "icon": "🔔",
>   "title": "Lorem Ipsum",
>   "text": "Lore  ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
> }
> ```
> 
> **Text and background Image**
> 
> ```
> {
>   "layout": 3,
>   "image": "http://url.to/image",
>   "text": "Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor"
> }
> ```
> 
> Image Only:
> 
> ```
> {
>   "layout": 4,
>   "image": "http://url.to/image"
> }
> ```

API implementation would be incredible for users to customize their experience crowdsourced, having more flexibility not only will make things easier for a sole developer but also more selling points for the device

lesongbang, 2024-10-14:
Nếu có thể tích hợp api như này thì thật tuyệt vời

lesongbang, 2024-10-18:
Chủ group ơi. Thêm vào HTTP custom text and images đi!

mungle, 2024-10-24:
Hey everyone!
It looks like you all liked the idea, as the post received a lot of positive reactions and suggestions for improvements. I was wondering if @GeekMagicClock has had a chance to see our request and start implementing it.

mo3head, 2026-01-22:
@GeekMagicClock people would love the API feature 👍

mungle, 2026-05-03:
>...I hope it could be finished in the next few weeks.

Hello! Updates about this project?