GeekMagicClock/smalltv #23: 오류

arron0720, 2024-03-09:
초기 설정 후 공장초기화 이 후 
192.168.4.1 접속시 

1.set your city first 
....
input your city location:  ( 185848                  )

save 저장안됨 
이후 진행 안됨 

[image]
[image]
[image]
[image]

chs0612, 2024-03-15:
i have same problem

ferrellsl, 2024-03-21:
The only way I've been able to successfully get my city to show up correctly on the main screen is to use the numeric code instead of the city name. Go to openweathermap.org and search for your city. Once you find it, you'll notice a number in the URL of your browser. Use that number instead. I live in Mesa, AZ USA and my code is: 5304391 as shown in the following screen shot.

[image]

arron0720, 2024-03-21:
도시 번호를 입력했도 동일 오류 발생 함

ferrellsl, 2024-03-21:
> 도시 번호를 입력했도 동일 오류 발생 함

Ah, I see.  Sorry to hear that.

GeekMagicClock, 2024-03-24:
> After initial setup Run factory initialization and then
> 
> 192.168.4.1 connection will not proceed next.
> 
> If you access the page below, no matter what code you put in, it will not proceed to the next step.
> 
> ++++++++++++++++++++++++++++++++++
> 
> 1. Set the city first
>    ....
>    Enter your city location: (185848)
> 
> Save. ++++++++++++++++++++++++++++++++++++
> 
> An error that can't be saved!
> 
> Due to the above situation, the product cannot be used!
> 
> Please give me an answer to solve this problem.
> 
> Thank you.
> 
> ![KakaoTalk_20240309_111121648_01](https://private-user-images.githubusercontent.com/162768177/311405748-9d7a3426-8362-4f89-a493-205a555bde00.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyODQ5MDMsIm5iZiI6MTcxMTI4NDYwMywicGF0aCI6Ii8xNjI3NjgxNzcvMzExNDA1NzQ4LTlkN2EzNDI2LTgzNjItNGY4OS1hNDkzLTIwNWE1NTViZGUwMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQxMjUwMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTE2Mjk5MjYyNGY0M2RjMTEyNGEzMjIyNWVmNWUzYTExNGRkYTViYTY3ODE5MDEzOGU1ZmEwNGFiODA5YjNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.n7ItVtkkxBZPMjRRehQpwLDqO_M-zrU_KPaZ-9MCbns) ![KakaoTalk_20240309_111121648_02](https://private-user-images.githubusercontent.com/162768177/311405750-b3b08923-7d31-433c-bf63-bb270ff3a675.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyODQ5MDMsIm5iZiI6MTcxMTI4NDYwMywicGF0aCI6Ii8xNjI3NjgxNzcvMzExNDA1NzUwLWIzYjA4OTIzLTdkMzEtNDMzYy1iZjYzLWJiMjcwZmYzYTY3NS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQxMjUwMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZTlhYWEwZDQ4MjQ5NDM0NGNhZjVjZjc3OTM3ZTYzYWU4YTdkMTRjMDY3MWMzODBmOWJiZTUzMjRhM2VkODZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.E3X2ydid6paAv1uyEtadkYe26RqPw1vnJNgH6h9lAJI)

Hi, seems you have update to a wrong firmware version, please try http://your_clock_ip/update to update it back.