GeekMagicClock/smalltv #24: Cannot downgrade version

B13afgggfg, 2024-03-18:
The link for downgrading from 3.1.4 to 3.0.33 does not work anymore. The smalltv now cannot function and the website keep appearing to be enter your location and not anyother option. The confirmation button also does not work.
Help
[image]

ferrellsl, 2024-03-21:
The only way I've been able to successfully get my city to show up correctly on the main screen is to use the numeric code instead of the city name.  Go to openweathermap.org and search for your city.  Once you find it, you'll notice a number in the URL of your browser.  Use that number instead.  I live in Mesa, AZ USA and my code is: 5304391 as shown in the following screen shot.

[image]

GeekMagicClock, 2024-03-24:
> The link for downgrading from 3.1.4 to 3.0.33 does not work anymore. The smalltv now cannot function and the website keep appearing to be enter your location and not anyother option. The confirmation button also does not work. Help ![Screenshot_20240318-222041_Chrome](https://private-user-images.githubusercontent.com/163859700/313713156-48cba1fa-f077-4897-bec8-a081fa4e2a76.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyODUwODMsIm5iZiI6MTcxMTI4NDc4MywicGF0aCI6Ii8xNjM4NTk3MDAvMzEzNzEzMTU2LTQ4Y2JhMWZhLWYwNzctNDg5Ny1iZWM4LWEwODFmYTRlMmE3Ni5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQxMjUzMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMDA4ZDg3ZThlYzc2YzU1OGNiYTE5N2M0Yjc2M2JhMzY5MGVkNmI5YmI1ODBiNmQwYmMyMDYzNmUyZDAwNDU4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9mdZOl-g6rgPRSOb4TcclN1nqLPt9EdQvNYDaDv_p-8)

Please downgrade it, visit this http://192.168.1.100/update