GeekMagicClock/smalltv #26: err key

fragolinux, 2024-03-23:
i updated to latest 3.1.4 from 3.1.1, all good, until i put in my newly generated weather api key (valid...), then i get "err key" on top, and it even stopped updating the time using the ntp server... removing the api key once put in is not possible, so what to do? to get at least the time working, i had to click the "update" button near ntp server, but after every reboot it's gone again... for now i just factory resetted the device, put in just my city ID and wifi, and NO api key, and it started working again... so, what to do? thanks

GeekMagicClock, 2024-03-24:
Hi, can you show your key here? The "err key" possibly means the key is invalid.

GeekMagicClock, 2024-03-28:
If that is a new generated api key, it may take hours to take effect.

closed 2024-10-09 by GeekMagicClock