GeekMagicClock/smalltv #30: Weather icon does not work propperly

DJ-SNAHS, 2024-04-09:
I have the latest firmware, but I can't view the weather icon. Is any way to view the weather icon correctly? Thanks
[image]

GeekMagicClock, 2024-04-11:
Hi, can you show me your settings of city? The city name should be displayed.
The gray circle means it is clear at night, when it comes to day time, the icon will be change to another one.

DJ-SNAHS, 2024-04-12:
> Hi, can you show me your settings of city? The city name should be displayed. The gray circle means it is clear at night, when it comes to day time, the icon will be change to another one.

Yes, the city name appeared well, but for privacy reason I deleted it.  I'm going to chek if the gray circle change, I changed the city  code to another code reference. Thanks for the help, I will keep you informed.

DJ-SNAHS, 2024-04-12:
Well I tested now, looks like the weather icon was freezed, I changed to another country and now it's working well I hope...  Thanks for the support.  My best greetings.

GeekMagicClock, 2024-04-12:
Thanks for udpate back~

closed 2024-04-12 by GeekMagicClock