GeekMagicClock/smalltv #34: Clock reboot loop after uploading custom GIF

Wolfeee18C, 2024-04-18:
After uploading a custom GIF, the clock seems like not loading the GIF and reboot itself. Then the clock start a reboot loop each time I plug it in. I've tried reboot, clean setting and even re update firmware but the file seems like stored in the clock and each time I choose "Upload your own GIF" it starts a new reboot loop.

Version: SmallTV V3.0.33

GeekMagicClock, 2024-04-18:
Hello, could you send that original file to me? I can test it.  ifengchao1314(at)gmail.com

Wolfeee18C, 2024-04-18:
Hello! Thanks for answering. I apologize and I just realized the file I have uploaded is a jpeg file, is there anything I can do to delete it?

GeekMagicClock, 2024-04-19:
> Hello! Thanks for answering. I apologize and I just realized the file I have uploaded is a jpeg file, is there anything I can do to delete it?

Yes, there's a way, try this:

1. Plug it off when the progress bar starts going, repeat this twice.
2. The third time when you power it on, it will get into AP mode, you can configure it again.
3. This time, uploading the correct gif file, will restore the device to normal state.

I'll try to fix this, at least, do not reboot.