GeekMagicClock/smalltv #38: Brick smalltv

ralerch, 2024-05-20:
Hello, 
I try upgrade smalltv to ultra version and I have some issues so I try upload firmware via serial port but after this firmware does not work, neither ultra nor normal.
No wifi, no image on screen.
Esp8266 is workig I can upload programs via arduino and this is working
How to bring it back to life?

AndreySSH, 2024-09-09:
Hi, guys. I also bricked my smalltv with esptool flashing. I tried to flash firmaware dump from repo (firmware_V3.0.33.bin) using esptool but device did not boot, only put on serial port:

v0007d9a0
~ld
rf cal sector: 1020
freq trace enable 0
rf[112] : 0a

Looks like firmware_V3.0.33.bin is only for OTA upgrade from web form and it could not be used as raw firmware. Could anybody provide full flash dump or repo maitainer provide full (not OTA) firmware version? Thanks.

ralerch, 2024-09-09:
> Hi, guys. I also bricked my smalltv with esptool flashing. I tried to flash firmaware dump from repo (firmware_V3.0.33.bin) using esptool but device did not boot, only put on serial port:
>
> v0007d9a0 [...]

[esp8266read.zip](https://github.com/user-attachments/files/16935479/esp8266read.zip)

My screen didn't work, but I have a dump from a friend, maybe it will help you

AndreySSH, 2024-09-09:
ralerch, thank you very much. I unbrick my smalltv. But I still no idea how firmware organized and what difference in hardware between smalltv with 3.0.X, 3.1.X and smalltv-ultra.