GeekMagicClock/smalltv #40: Error: Not found: /update

M4TZ3-op, 2024-09-06:
Hello
I want to update my smalltv but I can't access the update site.
When i type in http://ipadress/update I just get the error "Not found: /update".
How can i fix that?
The smalltv came with firmware version 2.5.2 as it shows on the boot screen

GeekMagicClock, 2024-09-11:
Hi, 2.5.2 is a copy and fake product, not GeekMagic brand, firmware can't be used.

fahimrahmanbooom, 2024-12-25:
@M4TZ3-op Hi, I purchased this same product from aliexpress but getting the same error. Did you find any way or other firmware?

andmt1, 2025-02-17:
> [@M4TZ3-op](https://github.com/M4TZ3-op) Hi, I purchased this same product from aliexpress but getting the same error, i understood that this is actually a copy product. Did you find any way or other firmware?

You can try this:
https://github.com/andmt1/files/blob/main/pre-built-firmware.zip

make a backup before flashing. Flashing via USB