GeekMagicClock/smalltv #41: Smart Weather clock license?

DanLysk, 2024-11-10:
I needed to restore the clock firmware. Now it's asking for a license key. 

[image]

DanLysk, 2025-01-22:
> Sounds like you have a non-official firmware from a AliExpress seller same as me. I was tricked into updating mine and have the same issue.
> I have a email address to get a license from off the MAC address but unsure what they are doing with this data and if they are tracking etc

Hey there! Thanks for reaching out. I finally did figure out my issue. I had one that was DOA. I had purchased it last year. I never could get anybody to give me a license key to make it work again. So I bought another one! And I told the seller that I received it and it was asking for a license key he asked me for the Mac address and gave me a license key. Now I have two of them working. I felt safe, giving them that MAC address as this is not a device that stores personal information or anything. I think it's safe.

I hope this helps.
~Dan 

Daniel Lysk
***@***.***
www.CentralArkansasVegan.org
www.SunsetFarmsSanctuary.org

Woo-Hwan, 2025-04-12:
license key. 178246F5CE79

Woo-Hwan, 2025-08-20:
license key. 178246F5CE79
________________________________
보낸 사람: devicerulez ***@***.***>
보낸 날짜: 2025년 8월 21일 목요일 00:57
받는 사람: GeekMagicClock/smalltv ***@***.***>
참조: Woo-Hwan ***@***.***>; Comment ***@***.***>
제목: Re: [GeekMagicClock/smalltv] Smart Weather clock license? (Issue #41)

[https://avatars.githubusercontent.com/u/227638011?s=20&v=4]devicerulez left a comment (GeekMagicClock/smalltv#41)<https://github.com/GeekMagicClock/smalltv/issues/41#issuecomment-3206980866>

Hi all, I need the license key for the following mac address:
FC:F5:C4:BC:1E:7A
Anyone can help me?
Thanx