GeekMagicClock/smalltv #42: Any chance for future updates?

ViBE-HU, 2024-11-22:
I opened many tickets here and got zero replies and the last firmware is already 1 year old which is a bummer. Most of my requests just polishing related. I don't need crypto and stock related stuff but would be good to see improvements. So what is going on?

GeekMagicClock, 2024-11-23:
Hello, smalltv is not going to add new features, only bugs will be fixed, we will spend more energy on new products, thanks.

closed 2024-11-23 by GeekMagicClock

ViBE-HU, 2024-11-23:
> Hello, smalltv is not going to add new features, only bugs will be fixed, we will spend more energy on new products, thanks.

Now this is kinda sad. Because it is not mentioned anywhere and this model is still selling on AliExpress and people still request and report here.

Also this is the simplest device. And as far as I know the only with multilang support. I understand that you cannot support all devices until the end of time. However for example my requests are basically not new features, but fixes/polisment of the existing.

https://github.com/GeekMagicClock/smalltv/issues/28
https://github.com/GeekMagicClock/smalltv/issues/33
https://github.com/GeekMagicClock/smalltv/issues/35
https://github.com/GeekMagicClock/smalltv/issues/36

These can be implemented in any models and as far as I know none of the newer models got these changes. So why should I or anyone choose and buy the other models if there is no guarantee for support and for basic fixes?

Not everybody wants to mine with a simple digital clock and I perfectly understand if you disagree with that, but you also refused to release the clock's source code which is a sad thing while people are still interested to help and keeping away the clock from become e-waste early.