GeekMagicClock/smalltv #43: uploading gif

Dahanael, 2024-11-28:
Hi !
i'm trying to upload gifs on the Model: SmallTV-Ultra,Version: Ultra-V9.0.25. Problem is the memory that is really short, even when i compress gif a lot. isnt there a way to extend the memory by using a local path instead of uploading it directly ? 
Prob a dumb question, sorry. But i can upload only one small and compressed gif, thats frustrating 
thanks !

GeekMagicClock, 2024-12-01:
Hi, total free space is about 1MB, if you delete all the gif/jpgs under the weather/photo options, please check.

Dahanael, 2024-12-28:
Yes, it's still not a lot of memory. My question was if there wasnt a way to take gifs from a local folder instead of having to upload it ? Because my device is plugged into my pc directly

GeekMagicClock, 2024-12-31:
No, If you want to stream the GIF file to the device, it's hard to do for this hardware.