GeekMagicClock/smalltv #51: How to flash with esptool

MennoLabs, 2025-06-28:
I acquired an older smalltv that came with firmware 2.5.7.
I tried to upload V3.0.33 to it using esptool.
The upload worked, but the device not.
Can I put the old 2.5.7 back?
Where to find it?
Or didn't I use the proper address or parameters to flash?
I used
esptool.py --chip esp8266 --port /dev/ttyUSB0 write_flash --flash_mode dio --flash_size detect 0x00 firmware_V3.0.33.bin 
Also tried address 0x1000

GeekMagicClock, 2025-06-29:
Hi, never released a version V2.5.7, that sounds like a fake geekmagic product.

Michail1, 2025-07-03:
You can always flash back to 2.5.7 if you kept a copy.
I do note that you can't simple read from another unit to be flash to the old unit.  (I'm working on this now, but on version 2.5.2.)

MennoLabs, 2025-07-04:
Yes, I am beating myself up that I didn't backup the old firmware first.

brxie, 2025-08-28:
I managed to flash mine Aliexpress unit as I mentioned here https://github.com/GeekMagicClock/smalltv/issues/38#issuecomment-3234066296